LumaTeam/Luma3DS #1680: Accidentally deleted some titles

m4tzzz, 2021-10-03:
So um, my sd card was getting full of storage and I was desperate in getting more space, so I went into titles and deleted some titles thinking it was gonna give me some storage space. Well after that, I connected my 3ds into my wi-fi and as soon as I exited out of the settings app. It just became a black screen and I have tried to update my luma3ds. What do I do?

Oreo639, 2021-10-03:
Try booting into recovery mode and then try updating. (you should get a message saying no updates are available, but it should resolve the issue)
https://www.3dbrew.org/wiki/Recovery_Mode

If you want to free up space on your SD card, just use the built in SD card management and if you do need to us FBI, keep in mind that all titles listed in red are titles installed on to the nand.

m4tzzz, 2021-10-03:
AHHHHHHH THANK U SO MUCH

closed 2022-03-09 by TuxSH