LumaTeam/Luma3DS #1681: N3DS Clock causes system lock-up in Senran Kagura 2: Deep Crimson

DarthVitrial, 2021-10-08:
**System model:**
New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

 11.15.0-47

**Entrypoint (How/what you're using to boot Luma3DS):**
FastBoot 3DS 1.2

**Luma3DS version:**

https://github.com/LumaTeam/Luma3DS/commit/b83a0ab
**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: (Yes )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (Yes )

Show NAND or user string in System Settings: ( Yes)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

In Senran Kagura 2: Deep Crimson, if you have Clock Speed set to 804MHz, the entire console will hardlock in the middle of stage 2-01: Hebijo's Loose Ends, when the game tries to transition to a mid-battle cutscene. The cutscene works properly if the clock is set to 268mhz. 
I've tested this five times on my end, the clock speed is the only setting that seems to cause this freeze. 

**Steps to reproduce:**

1. Enable N3DS Clock (L2 cache can be on or off, it doesn't matter)

2. Play through to chapter  2-01: Hebijo's Loose Ends. 

3. Play through the battle until a mid-battle loading screen appears (this is supposed to play a cutscene showing Homura becoming Crimson Homura).
4. The system will hang on the load screen until you force a shutdown. No crashdump is generated. 

**Dump file:**
No dump file is generated, the system just locks until you force it to turn off.

AuroraWright, 2021-10-08:
As far as I can remember this is one of the known games that have issues running at N3DS clocks, you can use Rosalina to switch the clock to 268 for this particular game. It's not Luma's fault

suppai, 2022-06-05:
Closing as this is a game issue, not an issue with Luma or overclocking itself.

closed 2022-06-05 by suppai