LumaTeam/Luma3DS #1689: Luma's crash handler doesn't handle manual crashes by an app

Loganh4005, 2021-11-15:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Causing a manual crash by an app doesn't get handled by Luma

**Steps to reproduce:**

1. Download the attached cia
2. [crash.standard.zip](https://github.com/LumaTeam/Luma3DS/files/7535535/crash.standard.zip)

2. Click crash

3. Wait a bit, then it will display that it crashed by the home menu

**Dump file:**

Oreo639, 2021-11-15:
Instead of uploading a random CIA, can you share the example code?

Loganh4005, 2021-11-15:
UnityEngine.N3DS.Debug.Crash("Boom!");

i used unity to make the cia file... idk the actual c code

suppai, 2022-06-05:
Closing, as it concerns official software/devkit tooling.

closed 2022-06-05 by suppai