LumaTeam/Luma3DS #1691: Suggestion: AGB_FIRM sleep: Simulate buttons on lid close

joemck, 2021-11-20:
Is it possible to get AGB_FIRM to simulate a button combination on lid close?

We have sleep patches for GBA games where you push one button combination to put the game to sleep, then push another button combo or open the lid and it wakes up. Is it possible to get AGB_FIRM to send the sleep button combo when you close the lid? Even just a single hardcoded button combination that can be turned on or off on the config menu would be great. The default one seems to be L+R+select, at least in the New Super Ultimate Injector, but we can set it to almost anything.

asdolo, 2022-03-13:
This is actually possible. AGB_FIRM already does that :)

Check this: https://www.3dbrew.org/wiki/3DS_Virtual_Console#Config

[image]

I already tested it and it works!

joemck, 2022-03-13:
Wth... Nintendo even cited not being able to get sleep mode to work properly as a reason they never released GBA VC games outside of the ambassador program, and added that notice on the bottom screen about sleep not working, and it turns out they had a complete fix that only needs a sleep patch to work on most games? Amazing.

I'm off to manually take apart, hex edit and rebuild my GBA VC injects then. I really wish NSUI set this field when building a GBA inject with the optional sleep patch.

I opened this ticket thinking AGB_FIRM needed patching to do this and that Luma could do it. Now my only issue is with NSUI, so I'm closing my ticket here.

closed 2022-03-13 by joemck

asdolo, 2022-03-13:
I'll add this feature to NSUI eventually. For now at least it's documented here for anyone 😛.