LumaTeam/Luma3DS #1692: Nintendo 2ds Arm11 (core 0) Problem

Slonkuhroni, 2021-12-02:
So i deleted some red files on fbi which cause the break but i don't know how to fix it and Im panicking Idk if have a custom theme effects it.
[image]

urherenow, 2021-12-02:
GitHub isn't a help forum. Google how to get into recovery mode, and perform a system update. Alternatively, boot into Godmode9, and restore your most recent backup of your nand.

Slonkuhroni, 2021-12-02:
How?

Slonkuhroni, 2021-12-02:
ohh

Slonkuhroni, 2021-12-02:
Umm another way?

Oreo639, 2021-12-03:
@Slonkuhroni Follow the instructions here:
https://www.3dbrew.org/wiki/Recovery_Mode

Then click update. You should get a message saying no updates are available, that's normal.

closed 2022-03-08 by TuxSH