LumaTeam/Luma3DS #1693: File system corruption leading to slow startup of TM++

jstolarek, 2021-12-03:
**System model:**  New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):** Sys 11.14.0-46E

**Entrypoint (How/what you're using to boot Luma3DS):** boot9strap

**Luma3DS version:** 10.2.1

**Luma3DS configuration/options:**

Screen brightness: 4

Splash: Off

Splash duration: 1

PIN lock: Off

New 3DS CPU: Off

--

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

I have TwilightMenu++ on my 3DS. When I installed TM++ it took about 10-15 seconds to start but I noticed that after a while startup time got significantly longer - around 50-60 seconds. I've spent hours trying to figure this out and it seems that the problem is caused by a minor corruption of the SD card file system. More specificaly, once the problem (slow startup of TM++) occurs, I can then take out the SD card, insert it into my computer and check the file system integrity. Since I'm on Linux here's what I get:

```
# dosfsck -a -w -v /dev/sda1
fsck.fat 4.2 (2021-01-31)
Checking we can access the last sector of the filesystem
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  65:01/00
  Not automatically fixing this.
Boot sector contents:
System ID "mkfs.fat"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
     32768 bytes per cluster
        64 reserved sectors
First FAT starts at byte 32768 (sector 64)
         2 FATs, 32 bit entries
  15630336 bytes per FAT (= 30528 sectors)
Root directory start at cluster 2 (arbitrary size)
Data area starts at byte 31293440 (sector 61120)
   3901157 data clusters (127833112576 bytes)
63 sectors/track, 255 heads
      2048 hidden sectors
 249735168 sectors total
Reclaiming unconnected clusters.
Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Checking free cluster summary.
Free cluster summary uninitialized (should be 2431429)
  Auto-setting.
/dev/sda1: 3151 files, 1469728/3901157 clusters
```

I believe the crucial bit here is the `Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.` part. The problem goes away after fixing the file system on the card, but it only goes away for a while (see "Steps to reproduce" below). It seems that as long as the file system doesn't get corrupted TwilightMenu++ starts in about 10-15 seconds. Once file system gets corrupted TwilightMenu++ takes almost a minute to start.

My SD card is SanDisk Extreme 128GB A2.

**Steps to reproduce:**

I don't have a slightest idea how to reproduce this consistently. This file system corruption just seems to happen randomly after a relatively short period of using my 3DS - if I fix it and spend an evening doing stuff (launching games, playing) it's almost guaranteed to happen again. Some hypotheses I tested and excluded (by excluded I mean it is not a *guaranteed* way of triggering the problem):

  * entering god9mode
  * dumping cartridge from god9mode
  * installing CIA files
  * putting console to sleep
  * launching TM++

One thing that crossed my mind was that perhaps the way Linux formats the card differs from the way Windows formats the card and this causes the problem to appear. I didn't test that hypotheses though.

Squall-Leonhart, 2021-12-09:
don't enable the TWM fat cache on SD cards larger than 32GB.

jstolarek, 2021-12-09:
Is that the "Save FAT table cache" option under "Games and Apps settings" in TM++? If so it is set to "No". I think other TM++ are also set to defaults, except clock format (24h) and Settings music (Off).

closed 2022-03-09 by TuxSH

jstolarek, 2022-03-09:
Has this been resolved?

jstolarek, 2022-03-18:
I got no reply so I'm asking again. Is this problem resolved in a newer version of Luma? I've seen no discussion here, no links to any PRs or commits so I would guess that the answer is "no", but if that's the case then why is this ticket closed? If you need me to provide more information or test something on my 3DS please let me know.

profi200, 2022-03-18:
It's not a Luma issue. Luma does not handle the fs in any way other than reading or writing the config on boot (or writing exception dumps). Everything else is Nintendo's own fs driver and it's known to have a few oddities like the dirty bit or not updating free clusters causing homebrew apps needing to scan the FAT on writing data sometimes slowing them down for a bit (takes longer on big cards).

There is nothing Luma can do about it.

jstolarek, 2022-03-18:
Got it, thanks!