LumaTeam/Luma3DS #1694: Crashes using Checkpoint Cheats

jah0147, 2021-12-04:
**System model:**
Old 3DS

**Entrypoint (How/what you're using to boot Luma3DS):**
[Boot9Strap]

**Luma3DS version:**
10.2.1 Stable

Screen brightness: ( max)

Splash: ( none)

Splash duration: ( )

PIN lock: ( none)

**Explanation of the issue:**
When running Cheats with any version of Checkpoint, my 3ds crashes, causing an exception error. Main issue is trying to run any cheat on Pokemon: Alpha Sapphire on v1.0 and 1.4. My 3ds is updated to the latest version. The crashed stopped when i downgraded to luma 9.1, but cheats then did not work. Crashing seems to happen more when trying to run cheats specifically on Alpha Sapphire

**Steps to reproduce:**

1. install version of checkpoint that includes cheats

2. install any version of pokemon: alpha sapphire

3. add cheats to game through checkpoint

4.  run cheats in game

**Dump file:**

Shadow322, 2021-12-04:
Quit cheating then it won't crash

closed 2022-01-23 by jah0147

oreo639, 2022-01-23:
To answer the question this is a duplicate of:
https://github.com/LumaTeam/Luma3DS/issues/1272

The reason this occurs is due to a race condition in the 3DS's kernel that makes it so that some cheats will randomly crash on some 3DSes:
https://github.com/LumaTeam/Luma3DS/issues/1272#issuecomment-507049149

Unfortunately this would require a significant amount of effort to fix and no one either has the knowledge or desire to fix it.

To quote the flagbrew server faq:
> Frequently Asked Questions - Checkpoint 4
> I'm crashing when using cheats with a Pokemon game. What should I do?
> This is due to a kernel bug. Luma's cheat engine would need to be redesigned to fix this, which would cause another bug elsewhere. As such, there is no solution to this problem at this time besides using another console.