LumaTeam/Luma3DS #1697: [Enhancement] Add Romfs Patching for applets

DraconicNEO, 2021-12-06:
Hello I'm trying to make a Romfs patch for a system applet but I quickly discovered that Romfs patches seem to be denied for some reason. Would it be possible to enable Romfs patching in applets in the future and if not why?

DraconicNEO, 2022-03-14:
> That would probably require a home menu patch. .code patching for applets is currently supported. romfs patching for the home menu will be supported in the next luma release: #1634

That's Just home menu patching though, I don't think it'll work for the other applets i.e. Miiverse, Notes, Browser, etc. or the one I was trying to edit which is a System Data Archive Note.

oreo639, 2022-03-15:
> That's Just home menu patching though, I don't think it'll work for the other applets i.e. Miiverse, Notes, Browser, etc.

True.

> or the one I was trying to edit which is a System Data Archive Note.

Yeah, that isn't possible due to how layerdfs is implemented in Luma. (layerdfs in luma works by patching the code that loads the romfs, so you can't use it with archives.)

DraconicNEO, 2022-03-16:
> > or the one I was trying to edit which is a System Data Archive Note.
> 
> Yeah, that isn't possible due to how layerdfs is implemented in Luma. (layerdfs in luma works by patching the application's code that loads the romfs, so it won't do anything with archives.)

That's kind of a shame.

cooolgamer, 2022-10-26:
> Hello I'm trying to make a Romfs patch for a system applet but I quickly discovered that Romfs patches seem to be denied for some reason. Would it be possible to enable Romfs patching in applets in the future and if not why?

I tried to remove the code that deny this and it crashes instead, that is why it deny it instead.
I have no idea if it's actually possible, I saw the enhancement tag was added so I guess it is?
It would be really great for custom keyboard because for now it just completely replace the original one (and is obviously not signed making it not working on OFW, and even system settings)...

DraconicNEO, 2023-04-07:
Implemented in v12

closed 2023-04-07 by DraconicNEO