LumaTeam/Luma3DS #1699: Compiling errors

huetrashme0947, 2021-12-19:
I'm trying to compile Luma3DS from source code. However, I'm getting a couple of errors:
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:89:5: error: unknown type name 'FS_SdMmcSpeedInfo'`
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:110:49: error: passing argument 1 of 'FSUSER_GetSdmcSpeedInfo' from incompatible pointer type [-Werror=incompatible-pointer-types]`
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:112:46: error: request for member 'sdClkCtrl' in something not a structure or union`
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:115:31: error: request for member 'highSpeedModeEnabled' in something not a structure or union`
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:115:54: error: 'SYSCLOCK_SDMMC' undeclared (first use in this function); did you mean 'SYSCLOCK_SOC'?`
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:118:49: error: passing argument 1 of 'FSUSER_GetNandSpeedInfo' from incompatible pointer type [-Werror=incompatible-pointer-types]`
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:120:46: error: request for member 'sdClkCtrl' in something not a structure or union`
`/home/heuhaufen/Luma3DS/sysmodules/rosalina/source/menus.c:123:31: error: request for member 'highSpeedModeEnabled' in something not a structure or union`

urherenow, 2021-12-19:
When did you install the dep libraries? Looks like your ctrulib is outdated...

huetrashme0947, 2021-12-19:
> When did you install the dkp libraries? Looks like your ctrulib is outdated...

Installed it yesterday.

> often, you need to compile and install ctrulib from source, instead of relying on dkp-pacman. Same with libnx for switch.

Thanks, I'll try that later.

Oreo639, 2021-12-20:
> often, you need to compile and install ctrulib from source, instead of relying on dkp-pacman.

This is only the case with luma as new features in luma may require new features to be added to libctru

> Same with libnx for switch.

Generally not.

urherenow, 2021-12-20:
Don't say "generally not" when you don't know what you're talking about. A new libnx is required with damn near every Atmosphere update, and is hardly ever updated on the dkp repos by that lime.

Oreo639, 2021-12-20:
> A new libnx is required with damn near every Atmosphere update

Sure, I was just trying to clarify that this is a Luma/Atmosphere specific thing and most software do not require pre-release versions of libctru/libnx. Sorry if I was being rude.

TuxSH, 2022-03-09:
libctru had a new release, so this is a non-issue

closed 2022-03-09 by TuxSH