LumaTeam/Luma3DS #1702: 3DS Becomes unresponsive after being closed for awhile

Techguy791, 2021-12-29:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.13.0-45U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS)**

Boot9Strap

**Luma3DS version:**

v10.2.1-e30cd7e2-2gxldr (Nanquitas Plugin Loader fork for CTGP-7)

**Luma3DS configuration/options:**

Default EmuNAND: No

Screen brightness: Auto

Splash: None

Splash duration: N/A

PIN lock: None

New 3DS CPU: No (only happens when running O3DS games AFAIK)

Enable loading external FIRMs and modules: Yes

Enable game patching: Yes

Show NAND or user string in System Settings: Yes

Show GBA boot screen in patched AGB_FIRM: Yes

Set developer UNITINFO: No

Disable Arm11 exception handlers: No

Enable Rosalina on SAFE_FIRM: No
--

**Explanation of the issue:**
I noticed that sometimes when I close my 3DS for more than ~5 minutes, the entire system will become responsive aside from the home and power buttons. I'm allowed to press the home button to go to the home menu, and I can bring up the power off screen, but the touchscreen nor any other button are not functional in this state. This usually fixes itself with a hard reboot. I at first thought this was an issue with the newest CTGP-7 version, as that's when I first encountered this bug. However, I later experienced this when playing New Super Mario Bros. 2 (digital, dumped from a cartridge).

**Steps to reproduce:**

1. Start a game

2. Close the system and wait

3. Upon reopening the system, all buttons (except for the home and power buttons) and the touchscreen will no longer work until the system is rebooted.

**Dump file:**
None, this does not result in a crash.

Oreo639, 2021-12-29:
Can you reproduce this issue with the Luma3DS (the non-pluginloader version)?

suppai, 2022-06-05:
Closing as a tech support question; the issue tracker is not for tech support, try Nintendo Homebrew's assistance channels (https://discord.gg/C29hYvh).

closed 2022-06-05 by suppai

MystiK791, 2022-06-20:
@suppai I didn't even remember this was still open, sorry!