LumaTeam/Luma3DS #1705: Rune Factory 4 Undub layered FS insanely slow

Nazosan, 2022-01-07:
**System model:**
New 3DS (non-XL)

**SysNAND version (+emu/redNAND version if applicable):**
11.10.0-43U SysNAND and 11.15.0-47U -- no emuNAND 

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
v10.2.1

**Luma3DS configuration/options:**
Default EmuNAND: (N/A)
Screen brightness: (2)
Splash: (Before)
Splash duration: (1)
PIN lock: (off)
New 3DS CPU: (L2) (I also tried off with no change)
Autoboot EmuNAND: (N/A)
Use EmuNAND FIRM if booting with R: (N/A)
Enable loading external FIRMs and modules: (off)
Enable game patching: (on)
Show NAND or user string in System Settings: (on)
Show GBA boot screen in patched AGB_FIRM: (off)
Set developer UNITINFO: (off)
Disable Arm11 exception handlers: (off)
Enable Rosalina on SAFE_FIRM: (off)

**Explanation of the issue:**
So I don't know if this is even something that would be worked on (assuming Luma3DS is still being worked on at all,) but it may be indicative of problems that could potentially show up in other things too.  I was going to try some minor modifications (just a couple of simple texture things, nothing in particular in mind) of Rune Factory 4's undub (0004000000766600) and I found out something interesting:  LayeredFS is completely nonworking for it.  If there is a romfs folder present in /luma/titles/0004000000766600 present at all -- even if it's completely empty -- the game takes ages to load.  The initial black screen after the 3DS logo takes a fair while before it shows the logo but then when the logo comes up it completely freezes the system for a very very long time (even home and power buttons are non-responsive during that time.)  When the game finally does load, everything that does any sort of data loading takes longer.  For instance, even just swinging a weapon/tool pauses a moment (I assume it's loading the sound or something.)  The first time you speak to someone the text is a bit slower even (though this is more subtle than the outright freeze for a moment when swinging a weapon or tool.)

I realize this is a modified version of the game and generally not going to be supported because of that alone (in fact, if I decide to continue to mess with it I'll later on try to find the exact changes to the original version of the game and just LayeredFS those changes to see how it goes) but I think whatever is going on likely points to an issue that could crop up elsewhere (for example, on future translations perhaps.)  I don't know all the specifics of the exact modifications, but from what I understand they actually swapped language files over the Japanese version of the game (the original attempts found that replacing the audio in the English version was more trouble than it was worth as I recall) and at one point one of the people who worked on the mod did say something about adding extra folders to the romfs -- but nothing specific.  

Of course it should already be clear, but without the layeredfs folder the undub runs perfectly fine with no hitches (or at least none I have ever noticed.)  It loads quickly, never freezes when swinging weapons/etc, and so on.  Simply deleting the romfs folder is enough to make it work fine again (leaving the 0004000000766600 folder seems to be ok.)

**Steps to reproduce:**

1.  Create /luma/titles/0004000000766600/romfs folder (can be empty or can have files)
2.  Play the undub of Rune Factory 4 (synthetic title ID 0004000000766600)

**Dump file:**
N/A (no crash)

AuroraWright, 2022-01-07:
Accessing the SD card is really slow on the 3DS sadly, at least the way it's done in LayeredFS. If I had to guess, issues like this one will greatly depend on the SD card IOPS (Samsung Evo cards are really good in that respect e.g., Sandisk stuff tends to not be) and the way it was formatted (larger FAT cluster size helps).

Nazosan, 2022-01-07:
I'm pretty sure that none of what you describe would mean the game completely freezing the entire system on startup for minutes and then freezing up momentarily in such an extreme way.  If LayeredFS were _that_ bad I'm not even sure it would have been implemented would it?  I can't tell you exactly what IOPS this card would have, but it is a class 10 (which, I know, is mostly about write speeds, but that means read speeds do meet at least that 10MB/s minimum too) and supports UHS-I (whic hI suppose the 3DS doesn't use.)  I admit it's a SanDisk, but not a bad one.  Now, that said, I have another game (Neko Atsume) with a LayeredFS implementation (a simple fan translation) and it does _not_ do this.  It doesn't freeze the entire system for several minutes on loading the game and most actions have no obvious delay.  (I haven't compared before and after on things like loading the shop or etc, so they may be slower, but at least functionally the game seems fine without obvious delays.)  I really don't think it should completely freeze the entire system for several minutes on starting a game simply by having an empty romfs folder surely?

TuxSH, 2025-04-08:
This is a limitation of LayeredFS. I'm not sure how the official SDK works but it's possible it has optimizations relative to RomFS access (application code gets a binary blob and is responsible for mounting it).