LumaTeam/Luma3DS #1707: When quitting AGB titles on EmuNAND, Luma redirects to SysNAND until running & quitting System Setting or TWL card/title

MelonGx, 2022-01-11:
Console: New3DS
Region: Japan
Version: 11.15.0-47J (both Sys and Emu)
Luma version: 10.2.1 and 7.1 a6dcce0b
Description: As topic.
AGB titles are all legit (officially system transferred from an ambassador 3DS, both Sys and Emu formatted then re-downloaded from official eShop)

Shadow322, 2022-01-11:
> Console: New3DS Region: Japan Version: 11.15.0-47J (both Sys and Emu) Luma version: 10.2.1 and 7.1 [a6dcce0](https://github.com/LumaTeam/Luma3DS/commit/a6dcce0b936d0b829db1bbc6df3d52e1e5a00b54) Description: As topic.
> 
>     * AGB titles are all legit (officially system transferred from an ambassador 3DS, both Sys and Emu formatted then re-downloaded from official eShop)
> 
>     * SysNAND and EmuNAND have different ID0.
> 
>     * BTW AGB titles in EmuNAND can load “AGB titles in SysNAND” ’s savedata.

This is a emunand problem that can't be fixed so deal with it or quit using emunand and the problem goes away

MelonGx, 2022-01-12:
> This is a emunand problem that can't be fixed so deal with it or quit using emunand and the problem goes away

Just to report that this issue exists.

closed 2022-01-12 by MelonGx