LumaTeam/Luma3DS #1712: ARM11 Error after update to 11.15

paulipauli130, 2022-01-19:
I turned on my 3ds again after a long time. My buddy then said he just had to update him. Boom: the "softbrick" was already there. My Luma3ds is on version 10.1.3. Well, in any case, I always get this ARM11 error now. I wanted to update Luma3ds to version 10.2.1 by replacing the boot.fiem and boot.3dsx on the SD card. 
But that didn't work either because I noticed that Luma3ds is loaded from the internal memory of the 3ds. In yellow font, the Luma3ds menu does not say: booted from SD Card but it says: Bootes from CTRNAND via B9S. Does anyone here have a fixed or the same problem?
[image]
[image]

TuxSH, 2022-03-08:
You haven't properly updated Luma3DS on the SD card, or your SD card is defective.

closed 2022-03-08 by TuxSH

zoracifer, 2022-05-17:
i have the same problem and i dont know what to do