LumaTeam/Luma3DS #1713: Arm11 Error after 3DS firmware update

Wattz56, 2022-01-19:
**System model:New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):11.15.0-47

]

**Entrypoint (How/what you're using to boot Luma3DS):**boot9strap

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**v10.2 stable

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: (+ )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (+ )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
I am running Luma V10.2 and went to update the system firmware to 11.15.0-47. After the update, the system refused to boot, giving me an ARM11 error. 

**Steps to reproduce:**

1. Update system

2. power on system

**Dump file:**

[crash_dump_00000006.zip](https://github.com/LumaTeam/Luma3DS/files/7898972/crash_dump_00000006.zip)

zamolobby, 2022-02-04:
ive updated to v10.2.1 and im still getting the error

TuxSH, 2022-03-09:
You haven't properly updated, or you have a defective sd card.

closed 2022-03-09 by TuxSH