LumaTeam/Luma3DS #1717: An error occurred (ErrDisp)

NyaakoYT, 2022-02-19:
Firmware: 11.14
Luma3DS build v10.2.1-b83a0abb (https://github.com/hax0kartik/luma-hourlies/releases/tag/246-luma3ds-b83a0ab)

Error type: generic

Process ID: 39
Process name: Iron16
Process title ID: 0x00040000001b4000

Address: 0x0020d4c4
Error code: 0xd900458b

This happened while trying to play Fire Emblem Echoes NA DLC: Siege of Zofia Castle and crashes upon skipping dialogue at the very start.

Shadow322, 2022-02-19:
> Firmware: 11.14 Luma3DS build v10.2.1-b83a0abb (https://github.com/hax0kartik/luma-hourlies/releases/tag/246-luma3ds-b83a0ab)
> 
> Error type: generic
> 
> Process ID: 39 Process name: Iron16 Process title ID: 0x00040000001b4000
> 
> Address: 0x0020d4c4 Error code: 0xd900458b
> 
> This happened while trying to play Fire Emblem Echoes NA DLC: Siege of Zofia Castle and crashes upon skipping dialogue at the very start.
> 
> I've already tried using Luma3DS from the official releases (https://github.com/LumaTeam/Luma3DS/releases/tag/v10.2.1), but still doesn't work.

This isnt a help forum go to gbatemp

TuxSH, 2022-03-09:
Corrupt file, you haven't pirated the title properly (assuming you did)

closed 2022-03-09 by TuxSH