LumaTeam/Luma3DS #1718: "The FIRM in /luma is invalid or corrupted." when you boot the Golden Sun II TLA GBA Game?

qamichelemassa5859, 2022-02-27:
When booting, the game gives an error:
```
An error has occurred:

The FIRM in /luma is invalid or corrupted.

Press any button to shutdown
```

Shadow322, 2022-02-27:
> 

This is not a Luma3DS problem post on gbatemp as this isn't a help forum

qamichelemassa5859, 2022-02-27:
> > 
> 
> This is not a Luma3DS problem post on gbatemp as this isn't a help forum

I tried enable arm11 exception handlers you will get a fatal error:
```
 An exception occurred
 
 Processor:       Arm9
 Exception type:  prefetch abort (svcBreak)
 
 R0     00000000       R1     00000000
 R2     00000000       R3     08000000
 R4     DDD6A445       R5     FFFF0080
 R6     00000000       R7     00000000
 R8     08000000       R9     00000000
 R10    FFFFFFFF       R11    00000000
 R12    00000000       SP     10000000
 LR     080AABF0       PC     08085951
 CPSR   60000030
 
 Press A to save the crash dump
 Press any other button to shutdown
```

closed 2022-03-08 by TuxSH