LumaTeam/Luma3DS #1722: An exception occured - bug CheckPoint

k-roline8, 2022-03-13:
Hello,

I would like to write to you because I have a little problem with my New 3DS XL. I don't know too much about "hacking" so my vocabulary may not be adapted... (and I am french, sorry for my bad english)

I installed Luma on my console, which was in version 11.15, thanks to the tutorial on YouTube. So far, no worries, it worked well.

I installed Luma on my console, which was in version 11.15, thanks to the tutorial on YouTube. So far, no worries, it worked well.

Nevertheless, I wanted to test CheckPoint for cheats and it turns out that it doesn't work at all. The app itself works well but when I access the Rosalia menu on the game, it crashes by indicating an error message (which I send you). No matter the game...

Would you know what this is due to?

Thanks !

[image]

NightScript370, 2022-03-13:
Hello there.
This is because of checkpoint itself, and not because of Luma3DS. Its best to take this to their department.
Nevertheless, I will tell you that its best you reset your homebrew environment. I fear that the reason you've been getting these errors are because of the specific versions of the apps placed in video guides' AIO packages.

I'm not sure on the policies emposed in this repo, but I believe it is fine to link to 3DS.hacks.guide

On their homepage, they have a section on what to do if you already have CFW, and said section will jump you to their Finalizing Setup page. Follow this to redo your setup with the official tools and you'll be able to use Checkpoint easily.

They also have a Discord Server linked on the top of every page for support. For real time support, please join there.

I hope this helps! 😊

closed 2022-03-23 by TuxSH

StonedPicco, 2023-06-14:
I have the same problem, what was the solution?