LumaTeam/Luma3DS #1726: Possible mistake in `swapFramebuffers`?

Coder-256, 2022-03-16:
I was just browsing through the Luma3DS source (curious how to draw to the GPU in practice), and I noticed the following function:

https://github.com/LumaTeam/Luma3DS/blob/bc6e14ada784ce93f5dbd030bfc557a6ba5f9852/arm11/source/main.c#L174-L179

According to gbatek, this seems to correspond to the "NEXT"/"Display Framebuffer" bits of PDCx_FRAMEBUF_SEL_ACK. However it looks like the top and bottom screen are both being updated based on the top screen's register, is this intentional?

profi200, 2022-03-17:
That looks like a copy & paste mistake to me. In practice it probably has no negative effect.

As for drawing something i would recommend [3ds-examples](https://github.com/devkitPro/3ds-examples/tree/master/graphics).

TuxSH, 2022-03-28:
Thanks a lot, I've fixed it

commit 90d6420f (2022-03-28): Fix typo in swapFramebuffer, closes #1726
closed 2022-03-28 by TuxSH