LumaTeam/Luma3DS #1732: Is there a way to switch the temperature display from Celsius to Fahrenheit?

ArguaBILL, 2022-03-24:
If there isn't, would it be possible to add the ability to switch the temperature display from Celsius to Fahrenheit?

Masamune3210, 2022-03-24:
Celsius is used for 99% of tech stuff when it comes to temperatures, fahrenheit wouldnt be really useful

eku, 2022-03-24:
Please don't use imperial units. SI rules the world.

Shadow322, 2022-03-24:
Yes add Fahrenheit

ELY3M, 2022-03-24:
yes, it is easy to convert C to F   I think it should be based on settings or config.  people should have choice on C or F.   
I would prefer F.

TuxSH, 2022-04-14:
No, there's is no way to switch units (yet, anyway), and:

1. The MCU itself reports the temperature in Celsius degrees; and Luma doesn't have any kind of internationalization support
2. 
[image]

closed 2022-04-14 by TuxSH