LumaTeam/Luma3DS #1735: Outdated info in Wiki section regarding upgrading

twocows, 2022-03-27:
https://github.com/LumaTeam/Luma3DS/wiki/Installation-and-upgrade
"You can manually update Luma3DS by repeating "Installation" above, but we recommend using [KunoichiZ's fork of Hamcha's LumaUpdate](https://github.com/KunoichiZ/lumaupdate/releases/latest)."

That repo is no longer maintained and currently doesn't work. Should probably replace with something about universal-updater or whatever solution is preferred currently.

TuxSH, 2022-03-28:
Good point. I need to update the wiki someday.

ShockwaveNN, 2022-03-29:
This issue is duplicate of my older issue, which has no response 

https://github.com/LumaTeam/Luma3DS/issues/1709

Chlorus, 2022-05-30:
This is still an issue. It's kind of funny to watch the maintainer grow increasingly frustrated that people keep opening issues on his repo: https://github.com/KunoichiZ/lumaupdate/issues/101#issuecomment-1071992125

ShockwaveNN, 2022-05-30:
@Chlorus I was ready to contact the maintainer about this, but all issues are blocked so I cannot comment in any way (

Shadow322, 2022-05-30:
> This is still an issue. It's kind of funny to watch the maintainer grow increasingly frustrated that people keep opening issues on his repo: [KunoichiZ/lumaupdate#101 (comment)](https://github.com/KunoichiZ/lumaupdate/issues/101#issuecomment-1071992125)

Has bot been an issue ever since Universal Updater replaced it

ShockwaveNN, 2022-05-31:
> Has not been an issue ever since Universal Updater replaced it

No, it is an issue since [Luma 3DSdocumentation](https://github.com/LumaTeam/Luma3DS/wiki/Installation-and-upgrade#upgrading) clearly states that you need to use LumaUpdate

Shadow322, 2022-05-31:
> > Has not been an issue ever since Universal Updater replaced it
> 
> No, it is an issue since [Luma3DS documentation](https://github.com/LumaTeam/Luma3DS/wiki/Installation-and-upgrade#upgrading) clearly states that you need to use LumaUpdate

https://github.com/LumaTeam/Luma3DS/issues/1735#issuecomment-1081054831

ShockwaveNN, 2022-05-31:
> https://github.com/LumaTeam/Luma3DS/issues/1735#issuecomment-1081054831

Good luck telling some inexperienced users that documentation is wrong but some comment is some-issue-you-need-to-find - contains correct info

TuxSH, 2022-06-03:
Fixed.

closed 2022-06-03 by TuxSH