LumaTeam/Luma3DS #1738: Game crashes in 3GX Edition

User9013, 2022-03-30:
Hello.
On my 3GX Loader Edition Luma3DS, when I load the Blank Plugin and start "Run for Money", the screen turns red and the game crashes.
The Luma and blank plugins used are as follows.

Luma3DS v10.3-f2e17d30-3gxldr
https://github.com/Nanquitas/Luma3DS/releases/tag/v10.3

CTRPluginFramework-BlankTemplate
https://github.com/PabloMK7/CTRPluginFramework-BlankTemplate/releases/tag/v0.7.2

Attached is a dump file of the crash.
Thank you in advance for your help.

[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/8380121/crash_dump_00000000.zip)

3DS: Old 3DS LL
SysNAND version: Sys 11.15.0-47J
Emu NAND: Not Used

Shadow322, 2022-03-30:
Not a Luma3DS problem use the proper github

User9013, 2022-03-31:
I will report this as a problem on the plugin side.
Thank you very much.

closed 2022-03-31 by User9013