LumaTeam/Luma3DS #1739: ARM9 crash error when trying to enter Homebrew Launcher

gvaldebenit, 2022-03-31:
**System model:**

Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.15.0-47U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

10.3 Release

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4 )

Splash: (Off)

Splash duration: (1)

PIN lock: (Off)

New 3DS CPU: ( )
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( x )

Show NAND or user string in System Settings: ( x )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

When trying to enter to homebrew launcher via injected into an app, ARM9 error appears. I double checked boot.3dsx is on SD Root, and tried with differents files and apps in case file is corrupted with no results.

**Steps to reproduce:**

1. Open an app to inject Homebrew Launcher

2. Inject Homebrew Launcher with Rosalina Menu

3. Close App then open again

4. Crash (ARM9 Data Abort).
 

**Dump file:**
[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/8385464/crash_dump_00000000.zip)

jtgrace51, 2022-04-03:
Same problem exactly. Newb of the entire homebrew 3ds community following [this guide here](https://3ds.hacks.guide/finalizing-setup) and getting the exact problem that @gvaldebenit is having. Can't get past this stage, don't know what to do, but I'm going to try an old version of Luma3ds and I'll update with results.

PabloMK7, 2022-04-05:
ARM9 errors are caused by a faulty or corrupted SD card. You will need to format the SD card to FAT32 again or replace it with another one.

impeeza, 2022-04-05:
use h2testw to test your SD card, could be a fake or defective one.

gvaldebenit, 2022-04-07:
Yeah, problem solved after a quick format to SD, seems like a corrupted file, but none of the games were affected. Thanks for the help.

closed 2022-04-07 by gvaldebenit