LumaTeam/Luma3DS #1742: arm9 data abort for tomodachi life

septichacker, 2022-04-11:
it only happens in tomodachi life, is it a bad save or a corrupted cart?
It happens when the 3rd item appears on the loading screen

luma: v10.3-f2e17d30-3gxldr
3dsxl: sys v11.15.0-47
[image]
E

hurda, 2022-04-11:
https://github.com/LumaTeam/Luma3DS/issues/1739#issuecomment-1089353996
>ARM9 errors are caused by a faulty or corrupted SD card. You will need to format the SD card to FAT32 again or replace it with another one.

septichacker, 2022-04-11:
so is it a corrupted save or is my whole sd card corrupted because i never get any other arm9 errors with anything

septichacker, 2022-04-12:
well, your suspisions were correct, my sd card soon then failed to read and had to be formatted.

septichacker, 2022-04-12:
issue resolved after a quick format :D

closed 2022-04-12 by septichacker