LumaTeam/Luma3DS #1743: Official 3GX plugin support?

Yrouel, 2022-04-19:
I wanted to try CTGP-7 (Mario Kart 7 mod) and it requires a fork of Luma3DS with the support for this type of plugins.

I also saw this closed pull request to add the feature on the main version (#1526 adding 3gx plugin support officially) and I was wondering why it wasn't actually implemented also as a way to focus the development efforts on one version instead of you know multiplying such effort.

So yeah consider this a feature request

Dimensional, 2022-04-20:
It's pretty good, though they still are working out the kinks for the plugin libraries. Some games will fail to work with their plugins, such as old SDK games like Pokemon Dream Radar, which I've personally tested.

suppai, 2022-06-05:
https://github.com/LumaTeam/Luma3DS/pull/1526 adding this feature was not accepted for merging.

closed 2022-06-05 by suppai