LumaTeam/Luma3DS #1745: Configuration screen opens...

t33st33r, 2022-04-29:
Hello,
I use both your original and Nanquitas' mod of Luma3DS.
Configuration screen opens every time I switch on booting from one payload to another. It opens no matter if select key is pressed or not, it ignores the the fact that the configuration file is present. The configuration screen reset all values every single time this happen.

Can you fix it? Please.

oreo639, 2022-04-29:
> it ignores the the fact that the configuration file is present

Luma checks if the configuration file is compatible with the current version and will create a new one if it is not. If you are switching between two versions of Luma with incompatible configuration files, you will be asked to update your settings.

t33st33r, 2022-04-29:
> 

Both my payloads are up to date (10.3 base). Why should not be compatible?

oreo639, 2022-04-29:
Probably because one adds options that don't exist in the other, which causes the size check to fail.

t33st33r, 2022-04-29:
OK thanks.
Now I need to find a way to contact Nanquitas. The project doesn't have section to post issues.

PabloMK7, 2022-04-30:
I'm maintaning the plugin loader branch. This issue will no longer happen after the next release, as configuration will be loaded from an ini file.

suppai, 2022-06-05:
No longer an issue as of v11.0.

closed 2022-06-05 by suppai