LumaTeam/Luma3DS #1746: ERROR: Weird/Crappy arm9 exception type prefetch abort after Formatting my SD card on 3ds?

subad31155, 2022-04-30:
When formatting my SD card, it gives an arm9 prefetch abort exception error.

PabloMK7, 2022-05-02:
English: The error means your SD card is BROKEN. Nobody can fix your error because your SD card is BROKEN. You need to buy a new SD card as it can not be repaired. Please stop stalking every single homebrew project with your errors.

Spanish: El error significa que su tarjeta SD está ROTA. Nadie puede corregir su error porque su tarjeta SD está ROTA. Debe comprar una nueva tarjeta SD ya que no se puede reparar. Por favor, deja de acosar cada proyecto homebrew con tus errores.

French: L'erreur signifie que votre carte SD est CASSÉE. Personne ne peut corriger votre erreur car votre carte SD est CASSÉE. Vous devez acheter une nouvelle carte SD car elle ne peut pas être réparée. S'il vous plaît, arrêtez de traquer chaque projet homebrew avec vos erreurs.

Portuguese: O erro significa que seu cartão SD está QUEBRADO. Ninguém pode corrigir seu erro porque seu cartão SD está QUEBRADO. Você precisa comprar um novo cartão SD, pois ele não pode ser reparado. Por favor, pare de perseguir cada projeto homebrew com seus erros.

German: Der Fehler bedeutet, dass Ihre SD-Karte DEFEKT ist. Niemand kann Ihren Fehler beheben, weil Ihre SD-Karte DEFEKT ist. Sie müssen eine neue SD-Karte kaufen, da sie nicht repariert werden kann. Bitte hör auf, jedes einzelne Homebrew-Projekt mit deinen Fehlern zu verfolgen.

Italian: L'errore significa che la tua scheda SD è ROTTA. Nessuno può correggere il tuo errore perché la tua scheda SD è ROTTA. È necessario acquistare una nuova scheda SD in quanto non può essere riparata. Per favore, smettila di perseguitare ogni singolo progetto homebrew con i tuoi errori.

Japanese: エラーは、SDカードが壊れていることを意味します。 SDカードが壊れているため、誰もエラーを修正できません。 修理できないので、新しいSDカードを購入する必要があります。 すべての自作プロジェクトをエラーでストーカーするのはやめてください。

f4arm4, 2022-05-03:
> English: The error means your SD card is BROKEN. Nobody can fix your error because your SD card is BROKEN. You need to buy a new SD card as it can not be repaired. Please stop stalking every single [...]

This is a question
????????????
Why?

Masamune3210, 2022-05-03:
Flash memory's lifespan isn't infinite. SD cards have a finite amount of work they will do before the flash inside becomes too degraded to work properly anymore and the card goes into read-only to protect from data loss

PabloMK7, 2022-05-03:
> This is a question
> ????????????
> Why?

This user just keeps reporting the same error to every single homebrew community out there. When we tell them that his SD is broken he just doesn't respond and stars over again reporting the error. If we ban them, then they create an alt account and do the same thing.

profi200, 2022-05-03:
inb4 f4arm4 is another alt. But seriously, these useless, gibberish issues start to get annoying.

Masamune3210, 2022-05-03:
Almost sounds like an intentional troll with that kind of tactic

Masamune3210, 2022-05-06:
Could always enforce a issue template with the new github stuff that way issues cant make it through that are just gibberish

Omega207, 2022-05-12:
Enforcing a issue template might not help, because the user might leave everything blank but the main topic, but it might help prevent this anyways if the user is a bot.

Masamune3210, 2022-05-12:
I mean with GitHubs enforced templates, you can set it up so that fields are required. Would almost kill bot posts and at least make it harder for people to make these kinds of issues

Omega207, 2022-05-13:
@Masamune3210 I see. I actually didn't know that; I'm quite a noob at GitHub. In that case, yes, that probably would help keep out any bots attempting to make nonsensical issues.

closed 2022-05-21 by TuxSH

BimRai, 2024-01-16:
For anyone who is stuck with this issue, just follow this **link**: https://3ds.hacks.guide/checking-for-cfw#what-to-do-next then choose your version of Luma. Just make sure to read carefully whilst following the steps. Your SD card is not damaged or broken whatever these guys are saying. Just make sure to follow the steps clearly because at this stage you already have Luma/Mod, you just need to install the rest correctly in order for the mod to proceed and complete. Think of it as you are half way there. I had the same issue but reading this felt irritating but I did some digging and managed to find a fix so to quit wasting time for any new visitors here, follow the link and it will be okay! :)

Masamune3210, 2024-01-16:
This has absolutely nothing to do with this issue.
1. Please make sure you actually are responding with something useful
2. Please do not comment on long-closed issues.