LumaTeam/Luma3DS #1749: 3DS crashes on startup

yoshiboyadvance, 2022-05-04:
**System model:**

New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.15-47E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v10.3 stable

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (After)

Splash duration: (5 seconds)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
The 3DS crashes on startup with an exception because of the process "act" (Title ID: 0004013000003802). The network light comes on, so my assumption is that it crashes when most of the operating system is already loaded.

**Steps to reproduce:**

1. Power on the 3DS.

2. Wait 10-ish seconds.

3. Presto!

**Dump file:**

[crash_dump_00000029.zip](https://github.com/LumaTeam/Luma3DS/files/8618469/crash_dump_00000029.zip)

yoshiboyadvance, 2022-05-12:
nvm it unborked itself after a week

suppai, 2022-06-08:
The issue tracker is not for tech support, please try Nintendo Homebrew (https://discord.gg/ZdqEhed).

closed 2022-06-08 by suppai