LumaTeam/Luma3DS #1754: Luma Updater Failure.

NickDeBeenAIE, 2022-05-24:
[image]

Download failed.

I am requesting a fix please.

oreo639, 2022-05-24:
Luma Updater is a third party project that has no relation to Luma3DS, and regardless this is the wrong repo to report bugs with it. (The correct repo is stated in the very screenshot you posted)

Also, as per:
https://github.com/KunoichiZ/lumaupdate#luma-updater
Luma Updater will not be receiving any new updates anyway.

oreo639, 2022-05-24:
No. Luma Updater is abandoned and will not receive updates.

The next release of Luma will add a system for copying itself to the nand when it has been updated.
https://github.com/LumaTeam/Luma3DS/commit/d798ff0efcac93af068bdfa0e01b056c725899ab

closed 2022-05-26 by ihaveamac