LumaTeam/Luma3DS #1755: 3ds dont start after installing

DarkNightSwe, 2022-05-30:
https://user-images.githubusercontent.com/106544109/171020695-67fc4523-90b1-43ee-b7ec-b9135d8079fa.mp4

When i try to start my 3ds it the notification led lights up in the color of white and turns of again. And this started to happen when i tried to download this.

impeeza, 2022-05-30:
this space is for bugs reports, please go to correct channels for support, on gbatemp there is a thread about this issue, you can search it and get the answer.

regards.

DarkNightSwe, 2022-06-02:
> this space is for bugs reports, please go to correct channels for support, on gbatemp there is a thread about this issue, you can search it and get the answer.
> 
> regards.

Could You Paste an Link to one that could help me?

impeeza, 2022-06-02:
Google.com -> discord luma3ds
Google.com -> gbatemp
gbatemp.net  -> search -> luma3ds or 3ds brick

Etc. You get the idea

TuxSH, 2022-06-03:
https://discord.gg/C29hYvh

Your video is very blurry too. I'm not sure if the LED is red or magenta:
- SD FIRM missing, CTRNAND FIRM corrupt: magenta
- SD FIRM corrupt, CTRNAND FIRM also corrupt: red

You've done something wrong when installing. Make sure boot.firm is properly located on the root of your SD card.

closed 2022-06-03 by TuxSH

rdistiso, 2023-02-20:
> https://discord.gg/C29hYvh
>
> Your video is very blurry too. I'm not sure if the LED is red or magenta:
>
> * SD FIRM missing, CTRNAND FIRM corrupt: magenta
> * SD FIRM corrupt, CTRNAND FIRM also corrupt: [...]

He said it was white