LumaTeam/Luma3DS #1756: Please make Luma's automatic CTRNAND update optional instead

Yrouel, 2022-06-05:
Given the points made in the comments about the commit (d798ff0efcac93af068bdfa0e01b056c725899ab) where the automatic copying to CTRNAND was introduced I was hoping the issue would have been addressed before the new release.

I don't think Luma should be copying itself willy nilly to CTRNAND, this operation should require user input to proceed. Not only one might prefer to keep a separate version of Luma on the NAND but might not want Luma at all or be using Fastboot instead.
Also the code that writes to the NAND might not necessarily be bug free neither now nor in the future should you continue to develop this project (case in point b79717e8485e093363bc57c5e02bc7af558e4133).

DeadSkullzJr, 2022-06-05:
You guys can make these tools back stuff up to my SD card for me, you can give extra features if you wish to do so, but **DO NOT** force automatic actions onto my system, I cannot stress this enough with developers. Custom firmware is catered to the user's preferences, not a standard set by a group of people who prefer things done the way they want. I respect your work on your software, but when it comes to solutions people rely/depend upon, you should make stuff like this optional for those who prefer either path.

I'll be honest with you, I don't understand the move with adding this feature into Luma. GodMode9 is already capable of doing this for people already, as harsh as this sounds, I feel like the direction of where Luma is going is making it to the point where you want people to have sole reliance on this rather than giving people the necessities and allowing them to have their preferences through other means, you reinvented the wheel, the original wasn't broken to begin with however. This isn't a necessary feature for Luma to have, GodMode9 has more sense in a feature like this because the role of that software is to literally be a NAND and SD manager, Luma isn't that, you shouldn't be making it something that it isn't, I don't want it setting anything to my NAND, it has no real business doing so.

I try my best to not allow any sort of homebrew to write to my NAND in any way unless it's absolutely necessary in some manner. Maybe to some of you reading this you won't understand my reasoning, you likely don't care what gets done to your NAND so long as what you want just works, but I just like keeping my system neatly managed and clean if I can help it, I prefer maintaining some sense of integrity with my NAND.

Before someone chimes in thinking I am being rude, I go about making my points straightforward, I don't beat around the bush, take all this how you will however.

profi200, 2022-06-05:
Personal opinion on this:
I agree mostly with the others. While i can see this being useful for noobs i think there should be a way to disable it or ask the user if they want to do this and recommend in the guide that they should accept. Power users can then deny if they want. I think the yes/no prompt is the best compromise.

Masamune3210, 2022-06-05:
Perhaps have it in the config file as a option to do it automatically or not? Or is it done before the config file is read

TuxSH, 2022-06-05:
To be clear, the copying to CTRNAND **only** happens once per upgrade. For example, let's say next version is v11.1; in that case it will overwrite `ctrnand:/boot.firm`. But if after that you go back to v11.0 (or drag&drop v11.0 again), it will not. This means you only have to replace boot.firm on CTRNAND every once in a while if you're not happy with that.

The harsh reality of things is that [many users don't even know what "the root of the SD card is"](https://images-ext-2.discordapp.net/external/Y_rt74vEPkXxfAmWeNiuB0cZq6luzpdTlt6rToPRo1g/https/i.imgur.com/QXHIvOz.jpg).

There has been a dozen issues opened here, and probably far more in the NH discord, because some people were incapable of simply replacing a single file on their SD card & had very, very outdated versions on their CTRNAND partition. If not for that, I wouldn't have bothered.

> you should make stuff like this optional

> GodMode9 is already capable of doing this for people already

I've been around for much longer than you are, and if anything **important stuff should be made mandatory**. I repeat, you cannot trust >90% of the userbase doing the right thing themselves (this may sound arrogant, but I've seen how assistance channels looked - a lot of people have hopelessly broken installs).

> I feel like the direction of where Luma is going is making it to the point where you want people to have sole reliance on this

This has always been the case, ever since I've started contributing in 2014~2015. What do you think stuff like langemu is? Seeing how Luma started monopolizing the CFW sphere shortly after that, it's clear enough to me that preferred that approach instead of the ""power user"" stuff.

Yrouel, 2022-06-05:
The 3GS Loader Edition (needed to play CTGP-7) has been updated to 11.0 with this change.
I want to try CTGP-7 but I don't want a third party fork on my NAND, too bad it'll copy itself.

What if

MegaPramuk, 2022-06-05:
> and if anything **important stuff should be made mandatory**

That's just humbug. Nothing should be made mandatory and definitely not something that involves the core of the system, the NAND, without the user's knowledge or even consent. 
That's just shady, even if the source code is free, it sure doesn't let the user use the software how he wants it but how the developer (you) wants.
Also, why do you even consider this mandatory? Why are you trying to fix what isn't broken?

> the official OS touches the NAND more often than you can imagine

So... because a man hits a child repeatedly means you can join in and beat up the child too?
Or am I misunderstanding what point you're trying to convey entirely?

> GM9 NAND backups aren't exact backups as it inserts essential console-unique data inside free space (unused padding bytes) in the backup.

And this... has nothing to do with the discussion, so why bother bringing that up?

Is there even a valid reason to not make it togglable at the least, like people suggested (either in the menu or in the config.ini)?
I mean, a valid reason that isn't that you just don't feel like it.

SCOTT0852, 2022-06-05:
> What if a fork like that changes the code that deals with the fs introducing some bug and f*ck up the NAND? ruh roh you're screwed shrug?
Then that's a bug with the fork and it's the fork's author's responsibility to fix it.
> So... because a man hits a child repeatedly means you can join in and beat up the child too?
It's one file write every few months. It's not going to kill your NAND.

TuxSH, 2022-06-05:
> I want to try CTGP-7 but I don't want a third party fork on my NAND, too bad it'll copy itself.

It will not copy itself if you copy upstream Luma first. And otherwise, virtually all Pablo's fork users will be happy with that.

> What if a fork like that changes the code that deals with the fs introducing some bug and f*ck up the NAND? ruh roh you're screwed shrug?

Either it's unmodified code from here, in which case I'd most likely have been made aware. If not, it's not my problem but the fork's. And it's the user's responsibility from not running malicious code. 

> Also, why do you even consider this mandatory? Why are you trying to fix what isn't broken?

Because I'm getting dozen of issues caused by people having an extremely old version of Luma (that they copied using a GM9 script when following the guide, but never upgraded again). Meaning this is an issue occurring with tens of thousand of people?

> Is there even a valid reason to not make it togglable at the least, like people suggested (either in the menu or in the config.ini)?

Because new release often increase the config file version number, causing the config file to be reset to default.

> So... because a man hits a child repeatedly means you can join in and beat up the child too?

Comparing core OS functionality to child abuse aren't we?

--- 

To me it looks like you have you have no fucking clue what you're talking about.

Also you're aware that Luma3DS ran from CTRNAND creates config files on the CTRNAND, and has been doing so since forever?

Yrouel, 2022-06-05:
So since it seems you won't add an option to disable this behavior any time soon what's the best way to disable it before compiling Luma?

I guess commenting out these two lines should do?

  ```
if (isSdMode)
      ok = fileCopy("0:/boot.firm", "1:/boot.firm", true, fileCopyBuffer, sizeof(fileCopyBuffer));
```

TuxSH, 2022-06-06:
Yes.

Masamune3210, 2022-06-06:
This all seems way overblown, Tux is right, CTRNAND gets modified almost constantly by the stock firmware alone, and this would vastly eliminate most of the user error issues that constantly show up here and almost every 3ds centred discord server I've seen. I've changed my mind, this is a good idea

DeadSkullzJr, 2022-06-06:
> To be clear, the copying to CTRNAND only happens once per upgrade. For example, let's say next version is v11.1; in that case it will overwrite ctrnand:/boot.firm. But if after that you go back to [...]

> The harsh reality of things is that [many users don't even know what "the root of the SD card is"](https://images-ext-2.discordapp.net/external/Y_rt74vEPkXxfAmWeNiuB0cZq6luzpdTlt6rToPRo1g/https/i.imgur.com/QXHIvOz.jpg).

> There has been a dozen issues opened here, and probably far more in the NH discord, because some people were incapable of simply replacing a single file on their SD card & had very, very outdated versions on their CTRNAND partition. If not for that, I wouldn't have bothered.

Sure, but holding people's hand doesn't teach them diddly squat. This isn't a good enough reason to force things onto people and label that as "helping" people. You want to know why most people know next to nothing about what they are doing, it's because you aren't teaching them the ropes (though that's not including those who intentionally want to be lazy and not learn). Maybe stop hand holding and start teaching, that's how you help people.

> I've been around for much longer than you are, and if anything important stuff should be made mandatory. I repeat, you cannot trust >90% of the userbase doing the right thing themselves (this may [...]

I've been around longer than you really know, sure maybe I didn't poke my head into the special homebrew dens of the internet when you did, but I have definitely been around the block to at least to know some things, this isn't my first time in the rodeo, I've been alive and capable of keeping tabs on things for many years, and have had discussions with people outside of these places regarding said content(s). You being around is a pointless point to make and really gives me the impression that you think you know what's best for everyone just because you walked around for a handful of years. You may have walked the walk for a bit, but you have no authority to say what's best for everyone's systems, only your own. Also, there are people out in the world that can learn everything you know, and then some potentially in much shorter time frames, so don't put your full bets on some years of experience giving you a higher playing field over others, not exactly a good ingredient in the recipe defining personality. Lastly, trust isn't yours to determine, that's for the people to find out on their own and or to give.

> This has always been the case, ever since I've started contributing in 2014~2015. What do you think stuff like langemu is? Seeing how Luma started monopolizing the CFW sphere shortly after that, it's clear enough to me that preferred that approach instead of the ""power user"" stuff.

This is mostly because options were never really given to people, when something new came out most developers were quick to abandon certain methods in favor of newer solutions, and while I can understand the approach of wanting to jump into newer concepts, sometimes what you already had worked just as good as what's newer, it just worked differently to reach the same goals. People want options, but when you force people to believe there is only one way, then yes it's pretty easy to say nearly everyone uses your solution when that's what is drilled into their heads. Your argument here is pointless anyways, because that's like telling everyone Windows should be used since nearly everyone uses it, yet Linux, Macintosh, etc. exist as a preference made out of choice, keyword being choice. Being a monopoly in a scene where the user is supposed to have the power to choose what they want sounds illogical. This is a bad mindset and a bad direction overall.

> The official OS touches the NAND more often than you can imagine, and by the way, GM9 NAND backups aren't exact backups as it inserts essential console-unique data inside free space (unused padding bytes) in the backup.

Ah yes, the predictable rebuttal points to cover your reasoning. First off, NAND backups are essential for various reasons, sure, maybe GodMode9 doesn't do that 100% correctly, but I did state that I don't want to mess with my NAND through means of homebrew **unless it's absolutely necessary**. I anticipate that something will get scuffed, it's not the point of being 100% scuff free, it's the point of controlling how much gets scuffed regardless of how big or small, complex or simple, it is. Also I know the NAND is written to constantly via stock means, I have recognized that for years, that isn't my point though, I clearly stated my point in the state of **homebrew**. You are very good at picking apart what you want out of what someone says, and I admire your ability to have counter arguments, but you are going to have to try harder if you plan to convince me that this feature is essential.

I still respect you and your work, but I recommend you reconsider your approach to things moving forward. I wouldn't want to see talent and potential wasted by a concept we already deal with by corporations and such, monopolies have never led to anything good and you know it as well as I do.

suppai, 2022-06-06:
Bullying non-power users out of the community by making things intentionally difficult when they don't need to be will *never* be the goal of Luma3DS or anyone on the Luma team. 3DS hacking is already "difficult" enough as it is, compared to the nearly one-click solutions on the Vita and similar. 

As for accusations of Luma having a "monopoly", I'm not sure how you can fault the Luma team for this; people use what works best for them. No one is forcing anyone to use Luma3DS; we did not hold the CakesFW, corbenik, etc. developers at gun point to get them to stop developing their own CFWs, they chose to do so on their own. If you want to create your own version of Luma3DS with this feature commented out you're free to, but this feature is something that will help 99.9% of users so your claiming to speak for them against evil devs "forcing" their way or the highway is laughable, and this feature has already received very positive public perception in NH and this very thread. 

As there are many reasons to have the auto-update feature and no valid reasons to not, this feature request will not be implemented.

closed 2022-06-06 by suppai