LumaTeam/Luma3DS #1757: Unable to update

Akira808State, 2022-06-07:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
Successfully downloaded the update, but I got a message saying there's an MD5 mismatch between the server and local file

**Steps to reproduce:**

1. Enter Luma updater

2. Select "Install stable version

**Dump file:**

oreo639, 2022-06-07:
Luma Updater is third party software and is not supported.

impeeza, 2022-06-07:
exactly, luma updater is another product, if you go to the github page, the product is no longer developed.  and if your take your time, look there are another similar issue here closed explaining the same.

closed 2022-06-08 by suppai