LumaTeam/Luma3DS #1760: Pokemon Rumble Blast Impossible to Play

0reo1, 2022-06-09:
**System model:**

New 3DS XL (Japanese LL Version)

**SysNAND version (+emu/redNAND version if applicable):**

Sys 11.15.0-47J
**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap

**Luma3DS version:**

Tested on v10.3 & v11.0

**Luma3DS configuration/options:**

Default EmuNAND: (N/A)

Screen brightness: ( 4)

Splash: (Before)

Splash duration: (3)

PIN lock: (None)

New 3DS CPU: (None)

--

Autoboot EmuNAND: (N/A)

Use EmuNAND FIRM if booting with R: (N/A)

Enable loading external FIRMs and modules: (Enabled)

Enable game patching: (Tested on & off)

Show NAND or user string in System Settings: (Enabled)

Show GBA boot screen in patched AGB_FIRM: (Disabled)

Set developer UNITINFO: (Disabled)

Disable Arm11 exception handlers: (Tested on & off)

Enable Rosalina on SAFE_FIRM: (Disabled)

--

**Explanation of the issue:**

Anytime I attempt to launch Pokemon Rumble Blast, regardless of whatever format it's installed from it crashes the system upon launching the game 

**Steps to reproduce:**

1. Launch Rumble Blast, and ARM11 error crashes system

2. If Disable Arm11 exception handlers is checked, the system still gives an error and cannot launch the game, regardless of means of installation
**Dump file:**

oreo639, 2022-06-09:
I can't reproduce this.

0reo1, 2022-06-09:
> 

Oh I attempted uploading the dump but I guess it didn't go through, will reupload it in a moment

closed 2022-06-18 by suppai