LumaTeam/Luma3DS #1762: Rosalina - Disable power to card only temporary

desolutionX, 2022-06-11:
If i disable the Power to the Gamecard slot its only temporary. It is enabled again every next time the 3DS boots up.

desolutionX, 2022-12-03:
The feature to toogle off the flashcards power by standard is implemented in the unofficial build by Nutez. By using TWLmenu++ the power will be toogled on and after return to Homemenu toogled off.
Thanx to Nutez for his fork

Masamune3210, 2022-12-04:
Thats cool, but forgive me for asking, what does it have to do with Luma? If its a unofficial fork, they should be where to talk about it, not here

desolutionX, 2022-12-04:
Sorry if i mix something up. It should be as information, for maybe using his fork as source for solving this.

tonyzehs, 2025-11-19:
+1. Please add a simple addition to remember and implement this setting on boot.