LumaTeam/Luma3DS #1765: ARM11 bootup screen error still appears even when Luma is updated

Nick1BunsYT, 2022-06-16:
I haven't used my 2DS XL in a while so I decided to update luma to get past the arm11 error screen, but when I updated luma, the error screen still appears, same text as always. either Nintendo already patched the new update or my 2ds is bricked. 
can someone explain to me why this is happening?

Nick1BunsYT, 2022-06-17:
note: the luma configuration screen says that it's on version 10.1.3 even though I updated it to version 11.0

Shadow322, 2022-06-17:
> note: the luma configuration screen says that it's on version 10.1.3 even though I updated it to version 11.0

What does it say in yellow at the bottom of the bottom screen?

Nick1BunsYT, 2022-06-17:
> > note: the luma configuration screen says that it's on version 10.1.3 even though I updated it to version 11.0
> 
> What does it say in yellow at the bottom of the bottom screen?

"Booted from CTRNAND via B9S"

Shadow322, 2022-06-17:
> > > note: the luma configuration screen says that it's on version 10.1.3 even though I updated it to version 11.0
> > 
> > 
> > What does it say in yellow at the bottom of the bottom screen?
> 
> "Booted from CTRNAND via B9S"

Then update your CTRNAND copy

Nick1BunsYT, 2022-06-17:
> > > > note: the luma configuration screen says that it's on version 10.1.3 even though I updated it to version 11.0
> > > 
> > > 
> > > What does it say in yellow at the bottom of the bottom screen?
> > 
> > 
> > "Booted from CTRNAND via B9S"
> 
> Then update your CTRNAND copy

I haven't done anything to my 3ds in a while so I have no idea where to get the update and how to update it

Shadow322, 2022-06-17:
> I haven't done anything to my 3ds in a while so I have no idea where to get the update and how to update it

If u re-download from Universal Updater it will automatically update the CTRNAND copy

Nick1BunsYT, 2022-06-17:
> If u re-download from Universal Updater it will automatically update the CTRNAND copy

I redownloaded universal updater onto my 3ds and put it into my CIA folder, but it didn't auto-update the ctrnand copy. does it still auto-update when booted up into the arm11 error/luma config screen?

suppai, 2022-06-18:
The issue tracker is not for tech support, only actual issues with Luma3DS. Try Nintendo Homebrew: https://discord.gg/C29hYvh.

closed 2022-06-18 by suppai