LumaTeam/Luma3DS #1766: I think it bricked my 2DS! HELP!!!

Xzz3DSjbluma, 2022-06-21:
So… I recently jail-braked my Nintendo 2DS and after a while, it started to get this error: 
[image]

This came without warning and I have tried to replace files like boot.firm and stuff but nothing worked. After about a week, I started YouTubing it and NONE of them worked or helped one bit. Now I really need help! Anything would be helpful and thank you!

wicorn29, 2022-06-21:
(this is my other account) Supper sorry. Might you know who can handle this?

wicorn29, 2022-06-21:
Oh, and I would also find this as a bug.

Shadow322, 2022-06-21:
> (this is my other account) Supper sorry. Might you know who can handle this?

U was already told @TuxSH close this please

suppai, 2022-06-22:
The issue tracker is not for tech support, only actual issues with Luma3DS. Try Nintendo Homebrew: https://discord.gg/C29hYvh.

closed 2022-06-22 by suppai