LumaTeam/Luma3DS #1768: Poochy & Yoshi's Woolly World - Crash When Closing Rosalina and When Finishing a Level

2nukes, 2022-06-21:
**System model:**
New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.15.0-47U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap

**Luma3DS version:**
Luma3DS 11.0 (but it was happening on 10.x as well)

**Luma3DS configuration/options:**
I've followed all the available troubleshooting guides and searched the net for info (I work in tech and I'm fairly technically adept). I understand that a lot of people encounter issues when they do things like use an out-of-region game, enable cheats, game patching, disabling exception handlers, etc. - I have done none of those things and haven't messed around with any obscure settings. Game patching off, cheats off, disable exception handler off (when it was an option in 10.x).

**Explanation of the issue:**
Issue #1645 was closed prematurely IMO. The reported crash is consistently reproducible and the other crashes are frequent.

- Crash when ending a level (not 100% of the time)
- Crash when closing Rosalina menu (every time)

[image]

**Steps to reproduce:**

**1. End of level crash (frequent but not 100% reproducible)**
Given I am a user with the latest updates of the 3DS system and Luma,

When I play P&YWW and complete a level,

Then when it would transition and load to the next menu/level, the game crashes with an ARM11 error.

**2. Theater Mode crash (likely related to above - also not 100%)**
Given I am a user with the latest updates of the 3DS system and Luma,

When I play P&YWW and navigate to the Theater and complete a quiz

Then when it would transition/load the next screen, the game crashes with an ARM11 error.

**Note:** Finishing a level or Theater quiz doesn't always crash, but it does so very frequently. I have not tested this fully yet, but it may be related to reading/writing save data.

**3. Rosalina menu crash**
Given I am a user with the latest updates of the 3DS system and Luma,

When I play P&YWW and open the Rosalina menu (L + Down + Select) and then press B

Then the game crashes with an ARM11 error.

**Dump file:**

[crash_dump_00000005.zip](https://github.com/LumaTeam/Luma3DS/files/8952982/crash_dump_00000005.zip)

Shadow322, 2022-06-21:
The horse is well decomposed by now guys sesh

2nukes, 2022-06-21:
> The horse is well decomposed by now guys sesh

?

suppai, 2022-06-22:
From your post on the the original issue: "I have tried different SD cards and CIAs and done every troubleshooting process I could find with no luck."

We don't support piracy. Please purchase your games from the eShop or as a cartridge to get a clean copy and issues like this would not happen. If you're still having this issue, investigate with NH (https://discord.gg/C29hYvh) and re-open this issue if they determine the issue to be anything besides a piracy issue.

closed 2022-06-22 by suppai

2nukes, 2022-06-22:
I own a copy of the game & It happened with my own dump, and then I tried others. Not pirating. I have already been in that Discord; they had no insights and referred back to Luma.

2nukes, 2022-06-22:
Please reopen issue.

Shadow322, 2022-06-22:
> So, this appears to be a locale issue with the game. I was able to reproduce these crashes using my USA copy of the game on my Japanese region N3DS, but after I used Luma Locale Emulator to set the locale to USA and EN, I was able to take screenshots, enter Rosalina, etc. without any crashes.

So not a Luma3DS bug after all

2nukes, 2022-06-22:
Tried that. I'm no longer getting the Rosalina crash, but I'm still getting crashes between levels and after Theater mode. I don't know why it would be a region issue; the first dump I tried was from my own game that matched my system's region (USA) (though I've been  experimenting with other dumps since then to see if my cart was somehow the issue).