LumaTeam/Luma3DS #1771: MD5 mismatch between server's and local file

SeleDreams, 2022-07-04:
**System model:** New 3DS XL
**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.14.0-46J
**Entrypoint (How/what you're using to boot Luma3DS):**
b9s
**Luma3DS version:**
10.2.1-c058e3f
**Explanation of the issue:**
It seems like the md5 hash of the latest 11.0 build might be incorrect as the download through the 3ds always results in a hash mismatch during the second check differentiating the local md5 and the server's md5
**Steps to reproduce:**

1. launch luma updater

2. update to the latest version

Shadow322, 2022-07-04:
Luma updater is dead and not of this team Universal Updater replaces it

SeleDreams, 2022-07-04:
> Luma updater is dead and not of this team Universal Updater replaces it

Ah I see, I went to this repository because luma updater linked to this one when it comes to bug reports

closed 2022-07-15 by suppai

Gumbraise, 2023-01-25:
So how did you do to update it ?

Shadow322, 2023-01-25:
> So how did you do to update it ?

U don't

strik3r93, 2023-02-01:
Yeah luma updater is broken, just update luma3DS manually

DRogueRonin, 2023-06-13:
How do I remove the Updater? I installed/updated it manually now but the updater still runs and checks and the only option I have is to quit