LumaTeam/Luma3DS #1779: Error using cheats (Luma3DS v11.0)

Gtasp1, 2022-07-22:
**System model:**

New Nintendo 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.15.0-47E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v11.0

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Off)

Splash duration: (No)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**

I recently bought a second hand New Nintendo 3DS XL.

I have installed Luma3DS (version 11.0), the console is in version 11.15.0-47E, and I try to use cheats using Checkpoint (version 3.7.4).

When I try to use any type of cheat in Pokémon Y or Pokémon Ultra Sun, like walk through walls or shiny Pokémon on, the console shows a black screen with the error Arm11.

What is happening? How can I fix it so I can use cheats without getting that error? Playing games naturally does not give any problem.

Thanks in advance.

**Steps to reproduce:**

1. Try any kind of game, like Pokémon and enter to the Rosalina menu (L + Down + Select).

2. Enter to the "Cheats" option and activate any of them.

3. Try to activate the cheat in game, and the black screen will appear.

[image]

**Dump file:**

[crash_dump_00000004.zip](https://github.com/LumaTeam/Luma3DS/files/9168613/crash_dump_00000004.zip)

Gtasp1, 2022-07-23:
I need to say that if I try to activate the "Enable game patching" option, the console shows me the same error.

suppai, 2022-07-29:
Issues like this are typically issues with the cheats or the installation of the game (often, pirated copies have issues). Please try the Nintendo Homebrew Discord.

closed 2022-07-29 by suppai

oreo639, 2022-07-29:
Duplicate of:
https://github.com/LumaTeam/Luma3DS/issues/1272

> There's a race condition in the kernel between the moments when:
>
>  +  CreateThread/Run attaches the KThread to a KDebugThread
>  +  The thread runs or any scenario involving thread->debugThread->x
>
> Basically you have to pray this doesn't happen.

Gtasp1, 2022-07-29:
._.

Shinysolgaleo, 2023-03-23:
hi, I understand it's been a while since this thread has been made but I'm having this problem as well and just wanted to know if you ever found a solution. Thank you.

Gtasp1, 2023-03-23:
> hi, I understand it's been a while since this thread has been made but I'm having this problem as well and just wanted to know if you ever found a solution. Thank you.

Hi, yes, the solution is easy:

If the console shows that black screen, it is because the cheat doesn't work. It happens with a lot of cheats in Pokémon, but the cheats in other games are fine. Sometimes, I don't let turn on the cheat all the game (for example, Shiny Pokémon in version X/Y), I turn on the cheat, return to game and one second after I turn off the cheat, and it works without black screen. Some cheats doesn't work and you need to search different ones on Internet, other cheats you need to use this treack of turn on/off in order to work, but not with all of them.

Sorry for my broken English. Regards.

Shinysolgaleo, 2023-03-23:
Hello, thank you for responding, I play ultra sun and when I turn on cheats, they will activate but then the screen will show that image from earlier in your post, but it doesn't happen so quickly then others it will, another reason I'm sure it does that is because a big part of the cheats have the 1.1 version when I'm playing on version 1.2. I don't know if they happen because of that but even using the few version 1.2 cheats it will show the black screen, if there is a way to get updated cheats for newer versions of these games then I will find out how. Thank you!