LumaTeam/Luma3DS #1780: Feature Request: Selecting Plugins?

TheGhostboi64, 2022-07-23:
I’m thinking of maybe add a feature where if you enable the plugin loader, you can select/deselect a plugin to use for your game or your system. Wonder if it’s ever possible to do that.

TheGhostboi64, 2022-07-23:
I don’t know how to add the enhancement label so that’s why it has no label

oreo639, 2022-07-23:
What "plugin loader" are you referring to? I assume you are talking about 3GX? If that is the case, this is the repo for Luma3DS which doesn't support those at all. If you have feature requests for a fork, you should ask there.

TheGhostboi64, 2022-07-23:
> What "plugin loader" are you referring to? I assume you are talking about 3GX? If that is the case, this is the repo for Luma3DS which doesn't support those at all. If you have feature requests for a fork, you should ask there.

It’s the 3gx plugin loader, I’ll just close jt

closed 2022-07-23 by TheGhostboi64