LumaTeam/Luma3DS #1781: Error with cheats

JFRR3, 2022-07-26:
**System model:**

[ Old 3DS]
[image]
[image]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Shadow322, 2022-07-26:
> Error with cheats, when I put The cheats And then I play in The game this error happens: arm 11 (core0) Type: abortion of imformation/data State of error: Traduction - section [image] [image]

This is not a Luma3DS problem

Gtasp1, 2022-07-28:
> > Error with cheats, when I put The cheats And then I play in The game this error happens: arm 11 (core0) Type: abortion of imformation/data State of error: Traduction - section [image] [image]
> 
> This is not a Luma3DS problem

I have the same problem. I think that this is a Luma3DS problem (the part of Rosalina's cheats) because the crash dump saves at  SD:/luma/etc etc etc.

Shadow322, 2022-07-28:
No this is not something Luma3DS will fox because it's not a Luma3DS problem @TuxSH please close this

Gtasp1, 2022-07-28:
> No this is not something Luma3DS will flx because it's not a Luma3DS problem @TuxSH please close this

So, if this is not something about Luma3DS, what is it?

[image]

Masamune3210, 2022-07-28:
Does it happen with every game or just one

suppai, 2022-07-29:
Issues like this are typically issues with the cheats or the installation of the game (often, pirated copies have issues).

closed 2022-07-29 by suppai