LumaTeam/Luma3DS #1783: Cheats in Luma3DS

Gtasp1, 2022-07-30:
I know that cheats doesn't work in Luma3DS, at least in version 11.0. Is there a way to solve that? Cheats work on emulators (for example, Citra or similars on Windows), but, why not in Nintendo 3DS? At least, there is another option in order to use cheats on 3DS?

Thank you in advance.

oreo639, 2022-07-30:
> I know that cheats doesn't work in Luma3DS

It does work.
There is a kernel race condition that can randomly occur on some consoles with some games (namely pokemon) but it should work fine most of the time.
Also, I can't reproduce your issue.

Omega207, 2022-08-06:
…What? I don't understand. Isn't using Checkpoint (via it's Sharkive database) technically using cheats? Or is there something I'm not understanding?
Sorry, just a question.

closed 2022-08-06 by suppai