LumaTeam/Luma3DS #1786: Ranged cache flush crash

csnikki, 2022-08-02:
**System model:**

Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**

11.15.0-47E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S

**Luma3DS version:**

v10.2.1

**Luma3DS configuration/options:**

Default EmuNAND: (NA)

Screen brightness: 4

Splash: Off

Splash duration: 1

PIN lock: Off

New 3DS CPU: (NA)

Autoboot EmuNAND: (NA)

Use EmuNAND FIRM if booting with R: (NA)

Enable loading external FIRMs and modules: No

Enable game patching: Yes

Show NAND or user string in System Settings: Yes

Show GBA boot screen in patched AGB_FIRM: No

Set developer UNITINFO: No

Disable Arm11 exception handlers: No

Enable Rosalina on SAFE_FIRM: (NA)

--

**Explanation of the issue:**

Custom syscalls for flushing cache with range crash the console. Syscalls for full flush work fine.

**Steps to reproduce:**

```c
svcFlushDataCacheRange(/* any address - size pair */);
svcInvalidateInstructionCacheRange(/* any address - size pair */);
```

**Dump file:**
[crash.zip](https://github.com/LumaTeam/Luma3DS/files/9240068/crash.zip)

PabloMK7, 2022-08-12:
Can you try with Luma v11.0, I remember I had this issue and I think @TuxSH fixed it.

csnikki, 2022-08-12:
I had tried before and had the same problem, but with the new codebase it seems to work, might have been a mistake on my end thanks.

closed 2022-08-12 by csnikki