LumaTeam/Luma3DS #1787: Investigate patching kernel bug

suppai, 2022-08-06:
There's a bug in the 3DS kernel that, while not related to Luma3DS, causes a considerable amount of end user inconvenience and issue reports to our repository: 

```
This seems to be a kernel bug (null deref), blame Nintendo I guess.

There's a race condition in the kernel between the moments when:

* CreateThread/Run attaches the KThread to a KDebugThread
* The thread runs or any scenario involving thread->debugThread->x

Basically you have to pray this doesn't happen.
```

It may be worth investigating if it's possible to patch this bug, as it is not likely Nintendo will do so.

PabloMK7, 2022-08-12:
Is that all the information you have related to that bug? Can you give more context on how to reproduce it? (What is your repository)

suppai, 2022-08-19:
@PabloMK7 This bug is one of the most common causes of random crashes, particularly with cheats, i.e. #1272.

OctoLinkYT, 2024-09-03:
Can confirm that it seems to happen commonly with Pokemon titles. It definitely wasn't always a thing as I had followed the exact same set of actions with no issues a few years ago.

(Just as a note, I do not condone cheating, I just use out-of-bounds cheats because I like to see the map out-of-bounds)