LumaTeam/Luma3DS #1792: Crash while opening Homebrew Launcher

superbeelavaman, 2022-08-25:
**System model:**

Old 2DS

**SysNAND version**

Sys 11.15.47U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

Was at 10.3 when the problem first occurred, updating to 11.0 didn't fix it

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

**Explanation of the issue:**

Whenever I try to open the Homebrew Launcher, either from Homebrew Launcher Loader or Download Play, it crashes. I can't install some homebrew software because of this.

**Steps to reproduce:**

1. Open app set to launch Homebrew Launcher

2. It Crashes after Splash Screen

**Dump file:**

[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/9428393/crash_dump_00000000.zip)

I removed entries from the Luma3ds Config section that weren't displaying for me in Luma3ds Config menu

oreo639, 2022-08-25:
Download the latest release of Luma from the releases section and make sure to copy all of the files including `boot.3dsx`.

superbeelavaman, 2022-08-25:
Thank you, this fixed it

closed 2022-08-26 by superbeelavaman