LumaTeam/Luma3DS #1793: Game/app patching wont work

csnikki, 2022-09-05:
**System model:**

Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**

11.16 sysnand

**Entrypoint (How/what you're using to boot Luma3DS):**

b9s

**Luma3DS version:**

11.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

I modified the exheader for the 3ds sound app (the one i use for homebrewing), to add a new IO region (NTRCARD); i wrote a program to automate that, so i'm pretty sure i got this right, ctrtool backs me up; i have the sdcard set accordingly (original code.bin and patched exheader.bin under `luma/titles/{title_id}`), when i launch my executable through the homebrew launcher, my 2ds segfaults when trying to access one of the IO registers. Building a cia with the same source code and the IO region in its rsf file seems to work fine.

**Steps to reproduce:**

1. Edit the 3ds sound app exheader to include IO region NTRCARD (0x1EC64000)
2. Place the exheader.bin under luma/titles/whatever
3. Start the homebrew launcher using the sound app as host
4. Execute a homebrew that access any register from NTRCARD

**Dump file:**

Shadow322, 2022-09-06:
just install the homeberew wrapper

csnikki, 2022-09-06:
Which one? The cia from yellows8 didn't work for me

Shadow322, 2022-09-06:
> Which one? The cia from yellows8 didn't work for me

This one

csnikki, 2022-09-06:
Doesn't work, same problem, it likely doesn't contain the definition for each IO region

profi200, 2022-09-06:
Why do you want to access the NTRCARD regs from userland?

csnikki, 2022-09-06:
@profi200 Don't see how that relates to the issue, but I'm fuzzing a card; I can use a custom CIA just fine, but the problem with luma patching remains

csnikki, 2022-09-06:
I have tested launching the homebrew from an application with the IO region set in the exheader; it crashes, so it looks like whatever system luma uses to launch boot.3dsx, it does not inherit the host capabilities. Is it possible for luma to always map all IO regions for every application? Kinda like how SVCs are unlocked

profi200, 2022-09-06:
It relates because i don't think userland apps are the right place for this. You will run into races with Process9 trying to access the same regs. I would try and modify for example GodMode9.

Again, i don't recommend it but if you insist use Luma's special memory mapping. You can access the regs like this:
`1u<<31 | 0x101640XX`

csnikki, 2022-09-07:
The special mapping works, thanks.

closed 2022-09-07 by csnikki