LumaTeam/Luma3DS #1794: my luma don't update

landy98, 2022-09-08:
I'm with luma in version 9.1, at first to update to 11.0, I just paste the files in the root of the sd, I did that and it continued in version 9.1, I searched a lot but didn't find anything, but when I open godmode9, I can boot boot.firm file from version 11.0 and it updates, but when I restart the console it goes back to version 9.1 and I have to repeat the process, godmode has an option to install boot.firm, but it has a message that it can brick the 3ds then I was afraid to try, I wanted to know what I do to be able to update luma normally

urherenow, 2022-09-08:
You've already installed it before. That's why it keeps booting 9.1. Now you either HAVE TO install the new version, or delete the old from NAND. Given your issue, I suggest installing the new version, because not understanding what you're doing while trying to delete something from the NAND, is how you you make a mistake and brick your console.

But to be extra careful, make a fresh backup of your NAND before doing anything at all. Also a backup of your SD contents, and keep them together. Make sure your OTP is backed up as well.

landy98, 2022-09-08:
I found out what the problem was, I saw that it appears boot from sd via firm0 instead of via b9s, I saw a tutorial on how to install b9s and managed to update luma to 11.0 in the standard way

closed 2022-09-08 by landy98

urherenow, 2022-09-08:
firm0 is part of your nand, not sd 🤦‍♂️

landy98, 2022-09-08:
I don't know much about luma, I knew that it updated when I put the files in the root of the sd, I had it and it didn't update, and I didn't know why or what I had to do