LumaTeam/Luma3DS #1795: ARM11 (core 1) when run Virtual Console

kevinka999, 2022-09-09:
I've being trying to play Pokemon Crystal but I get `ARM11 (core 1)` error when I run it.

I've already tried to fix it: 
Update the luma to latest (tried v11.0 and v10.3) and reinstall BS9 by using SafeB9SInstaller.

I saw some similar error here and on youtube but it seems be a different one, here is the dump file from it.
[dump_file.zip](https://github.com/LumaTeam/Luma3DS/files/9537872/dump_file.zip)

**3ds Firmware: 11.15.0-47U
Luma: v11.0
B9S: v1.4**

kevinka999, 2022-09-09:
I found a solution and actually is pretty simple.
For who is facing the same issue, follow these steps behind:
**Go to FBI -> Titles -> Game which is causing the error -> Import seed**

I really would like to understand the issue, but doesnt matter now because I can play xD

closed 2022-09-09 by kevinka999

Masamune3210, 2022-09-10:
Basically, later in the 3DS lifetime, N added another layer of encryption to titles as a way of handling pre-launch sales. They would let you buy the game and install it but wouldn't release the seed until the game was officially launched. The eshop is what handled this, but if you don't get your game through the eshop, you never get the seed, and the system eats sand when it tries to launch the title because it is still encrypted. Importing the seed installs the seed for that title into the systems database so the title can be properly decrypted and launched