LumaTeam/Luma3DS #1796: Luma3DS graphics ghosting when "Game patching" is enabled

AudioDuplication, 2022-09-14:
**System model:**
New 3DS XL (Maybe relevant: TOP IPS, BOTTOM TN)

**SysNAND version (+emu/redNAND version if applicable):**
11.15.0-47E SysNAND, 11.15.0-47E EmuNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Booted from SD via B9S

**Luma3DS version:**
v11.0

**Luma3DS configuration/options:** to reproduce my problem

Default EmuNAND: (1)
Screen brightness: (4)
Splash: (Off)
Splash duration: (1000)
PIN lock: (Off)
New 3DS CPU: (off)
--

Autoboot EmuNAND: (x)
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: (x)
Show NAND or user string in System Settings: (x)
Show GBA boot screen in patched AGB_FIRM: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )
Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
When I set "Enable game patching" to enabled. And open the Luma3DS configuration screen (holding select while booting), or use a splash screen, parts of splash/configscreen stay as a sort of pixel-retention/ghosting on the TOP screen (bottom is unaffected).
It will be visible on the homescreen, and in-game, and will very slowly fade out slightly but never entirely.

If I boot with "Enable game patching" off, then this issue does NOT occur.

The issue also does NOT occur when I take out my SD card and set "Enable game patching" to enabled.

I tested this on my New 2DS XL with 2 TN screens, the issue does not occcur, but it also does not have EmuNAND installed on the SD.

**Steps to reproduce:**
1. Have EmuNAND, and "Enable game patching" ON.
May or may not be necessary: Have New2DSXL with top TN screen.

2. Hold SELECT while booting, then press start to boot into 3DS

3. Either EmuNAND or SysNAND doesnt matter.

4. check the edges and corners of the screen for a ghost of what was on the screen during splash or the Luma settings screen.

[image]

PabloMK7, 2022-09-27:
Does this happen with Luma 10.3?

suppai, 2022-10-26:
Duplicate: #1396

closed 2022-10-26 by suppai