LumaTeam/Luma3DS #1798: There's a crash dump every time that I try to use a cheat

prglt, 2022-09-20:
Hello everyone, I wanted to use cheats on Pokemon Omega Ruby but when I try to use a cheat there's a crash dump, I have the current Luma version and I tried to use the last one before that. I don't know where the problem is so here's the image and I hope that it can be fixed. Thanks for the help!
[image]

Shadow322, 2022-09-22:
> Hello everyone, I wanted to use cheats on Pokemon Omega Ruby but when I try to use a cheat there's a crash dump, I have the current Luma version and I tried to use the last one before that. I don't know where the problem is and I hope that it can be fixed. the crash dump says: an exception occurred processor: Arm11 (core 0) Exception type: data abort Fault status: Translation - Section Current process: sango-1 (000400000011C400) Also, the console is upgraded to the last version so I suppose that's not the problem. Thanks for the help! [image]

dont cheat and it wont crash

prglt, 2022-09-22:
Is there a way to cheat and don't make It crash?

> dont cheat and it wont crash

Is there a way to cheat without making the game crash?

Shadow322, 2022-09-22:
> Is there a way to cheat and don't make It crash?
>
>
> Is there a way to cheat without making the game crash?

Nope

joseowo, 2022-09-23:
i think try using a alternative cheat code it can maybe fix your problem.

closed 2022-10-26 by suppai