LumaTeam/Luma3DS #1799: Screen brightness

dafish7, 2022-09-21:
So basically I opened Rosalina menu to change the screen brightness from 126 to maybe 40, but when I go to change the values, it doesn't change. 

N3ds with Luma v 11.0

Shadow322, 2022-09-22:
> So basically I opened Rosalina menu to change the screen brightness from 126 to maybe 40, but when I go to change the values, it doesn't change.
> 
> N3ds with Luma v 11.0

changes here just fine n3ds

BondarenkoArtur, 2022-09-27:
@dafish7 try disable auto-brightness if it's enabled

PorterBytes, 2022-10-01:
I made another suggestion quite some time ago (0 comments, boo), but it appears that the brightness resets every time you close the 3DS lid, where as ideally it would stick. Maybe that's what's happening here? Mine definitely changes (after you press A, THEN left or right, which isn't 100% intuitive).

closed 2022-12-04 by suppai